lab11/socitrack #5: Software bugs

abiri, 2019-01-14:
Currently known are the following high-priority bugs:

- [] DecaWave becomes unresponsive (despite hard and software resets) after running for multiple hours
- [] Neighbour Re-discovery starts the DecaWave in a mode where it cannot receive anymore
- [] Timestamps (presumably) observe a buffer overflow after several hours and are not reliable anymore thereafter

abiri, 2019-01-14:
- [ ] Node can be scheduled, but does not receive any valid replies from the master (can only occurs directly when starting the device, but remains stuck in this mode)

hedgecrw, 2020-09-03:
All remaining bugs in this thread have been addressed with the most recent code commit.

closed 2020-09-03 by hedgecrw