lab11/socitrack #7: Hardware issues

abiri, 2019-10-09:
**RTC related:**
- [ ] RTC battery discharges itself within multiple hours of the MCU being turned off
- [ ] RTC battery recharges itself without trickle charge being enabled on the RTC
- [ ] RTC nIRQU1 always remains low (does not correctly become high with pull-up as instructed by the nRF)
- [ ] Increase pull-up resistor values to decrease potential leakage (especially to pins which are connected to the MCU).

** SD card related:**
- [ ] When the SD card is turned off, SD chip select as well as MISO and MOSI are high, hence running current through the three pull-ups (10k) which results in 1mA of leakage. Increase pull-up resistor values

abiri, 2019-10-09:
**Solution:** Remove R17 and R18 as pull-ups, as they drain the battery while being connected to the MCU. The lines should be pull-ed up directly by the MCU.

abiri, 2019-10-09:
**Partial solution:** Connect EXTI to GND.

abiri, 2019-11-19:
**Partial solution:** Fix BOM for R11 to correct 1.41k value (instead of 10k)

volgy, 2019-11-19:
I am afraid that just removing R17 and R18 is not a proper solution. Yes, the MCU has pull-ups, but when it is not powered (the main battery switch is turned off), these nets will float. The datasheet for AB1815-T3 specifically warns against this. If WDI and EXTI are not used by the application, you may want to tie these directly to ground or  VBAT_RTC (with no connection to the MCU).

abiri, 2019-11-19:
Fair point; to still have them available at later times (e.g. in case one would want to use the Watchdog), I suggest pulling it to ground over some Pull-Down resistors. I will need to double-check whether there was a reason why I was not doing this in the first place.

volgy, 2019-11-19:
Yes, this (pull-downs) seems to be a good compromise.

ppannuto, 2020-07-05:
Revision G is shipped!

closed 2020-07-05 by ppannuto