lab11/socitrack #8: SW: Pin reconfigurations for Rev G

ppannuto, 2020-07-05:
Some of the pins for the nrf updated from Rev F to Rev G. I tried to make sure that nothing _bad_ would happen if old rev software runs, but the one risk is the `BUZZER_GATE`, which is floating and might burn out the speaker if not pulled off. That was an unconnected pin before, so it's safe to change that in the Rev F firmware as well such that the speaker is simply always controlled.

**Revision G:**
[image]

**Revision F:**
[image]

---

## Changed Pin List

(You should probably double-check me, just in case :) )

 - [ ] USB pins now connected
 - [ ] `LED_NRF_GREEN` `P0.06` -> `P1.05`
 - [ ] `DBG_RX` removed (was FTDI UART)
 - [ ] `DBG_TX` removed (was FTDI UART)
 - [ ] `!SD_DETECT` `P1.00` -> `P0.11`
 - [ ] `LED_NRF_BLUE` `P0.04` -> `P1.07`
 - [ ] `LED_NRF_RED` `P0.05` -> `P1.04`
 - [ ] (new) `!CHG` `P1.02` -- "Low (FET on) indicates charging and OpenDrain (FET off) indicates no Charging or Charge complete."
 - [ ] (new) `!PG` `PXXX` -- "Low (FET on) indicates the input voltage is above UVLO and the OUT (battery) voltage."
    - From https://www.ti.com/lit/ds/symlink/bq24040.pdf?ts=1593963502813&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FBQ24040
 - [ ] `!FT_RST` removed (was FTDI reset signal)
 - [ ] (new) `BUZZER_GATE` `P1.15` -- Drives a PFET that drives a buzzer. **Must be pulled high to default buzzer off.** This was intentionally put on an unconnected pin in prior revisions, so all firmware should drive this pin to ensure it isn't accidentally floated. Should be driven with a PWM signal to drive buzzer.
 - [ ] (new) `QI_TS_CTRL` `P0.25` -- `TS/CTRL` pin of [bq51013B](https://www.ti.com/lit/ds/symlink/bq51013b.pdf?HQS=TI-null-null-digikeymode-df-pf-null-wwe&ts=1593963941247), see §7.3.14 for details. Used to signal charge complete to wireless charger.

ppannuto, 2020-07-05:
Buzzer details:

[image]

http://www.puiaudio.com/pdf/SMT-0440-T-2-R.pdf

hedgecrw, 2020-09-03:
All changes implemented with latest push

closed 2020-09-03 by hedgecrw